Miceuz/i2c-moisture-sensor #1: Can it be used outdoor ?

txaj, 2015-06-28:
Or even buried 5-6 cm underground ? 

(I'm sorry to open an issue to ask a question but I couldn't find any way to contact you without registering on tindie)

Miceuz, 2015-06-29:
You would want to  cover in additional protection the place where cable 
is soldered to the sensor, some people are putting a tube over the 
electronics and pouring epoxy or silicone into it.

Regarding burying it underground - you would have to test this yourself 
;) The answer would be "probabaly" after you pot it with above method.

closed 2015-06-29 by Miceuz

dedekcz, 2020-04-16:
@Miceuz Nice project, Do You have any recommendation which epoxy is right one? is there any request for conductivity properties?

Miceuz, 2020-04-16:
You will get different reading with different kind of epoxy, but in general it does not matter al long as you calibrate after potting, make sure it's just non-conductive one, having no metal inside.