Miceuz/i2c-moisture-sensor #2: strange reading on the light sensor & bugs in example code??

krikk, 2015-11-06:
hi!

i got your sensor running with a raspberry pi and after a lot of research and tries i got very good readings.. my raspberry pi code:

    #!/usr/bin/python

    #https://github.com/adafruit/Adafruit-Raspberry-Pi-Python-Code/tree/master/Adafruit_I2C
    from Adafruit_I2C import Adafruit_I2C
    from time import sleep, strftime
    from datetime import datetime
    deviceAddr = 0x20

    i2c = Adafruit_I2C( deviceAddr, -1, False )

    #to change adress
    #i2c.write8( 1, 0x22 )

    #reset sensor, we need this otherwise i get inconsistent light reading in the dark...
    i2c.write8( deviceAddr, 0x06 )
    sleep(5)

    i2c.write8(deviceAddr, 3)
    sleep(3)
    light = i2c.readU16(4, False)
    temp = i2c.readS16(5, False)/float(10)
    moisture = i2c.readU16(0, False)
    print str(temp) + ":" + str(moisture) + ":" + str(light)

... only strange thing i noted there was that i had to reset the sensor always before reading values, because otherwise i got strange readings from the light sensor after a few reads... have no clue why... buts that ok with me.

but now i got a shiny new arduino uno and i tried to run your sensor with it, and once again i get wrong readings from the light sensor, e.g. i get a negative reading when measuring in a dark room

i used the provided example code from github a adapted it:

    void loop(){
      
        //Reset i2c moisture sensor
        Serial.println("Reset i2c moisture sensor");
        writeI2CRegister8bit(0x20, 6); //reset
        delay(5000);

        writeI2CRegister8bit(0x20, 3); //request light measurement 
        delay(3000);
        int light = readI2CRegister16bit(0x20, 4); //read light register
        int moisture = readI2CRegister16bit(0x20, 0); //read capacitance register
        float temp = readI2CRegister16bit(0x20, 5)/(float)10; //temperature register
        char tempstr[15];
        dtostrf(temp,4, 1, tempstr);

        Serial.print("moisture: ");
        Serial.print(moisture); 
        Serial.print(", Temp: ");

        Serial.print(tempstr); 
        Serial.print(", Light: ");
        Serial.println(light); 
        delay(2000);
    }

    void writeI2CRegister8bit(int addr, int value) {
        Wire.beginTransmission(addr);
        Wire.write(value);
        Wire.endTransmission();
    }

    unsigned int readI2CRegister16bit(int addr, int reg) {
        Wire.beginTransmission(addr);
        Wire.write(reg);
        Wire.endTransmission();
        delay(20);
        Wire.requestFrom(addr, 2);
        unsigned int t = Wire.read() << 8;
        t = t | Wire.read();
        return t;
    }   

 and i noted a few strange things in it:

    unsigned int readI2CRegister16bit(int addr, int reg) {
    ...
    }

why unsigned ?? ...what about negative temperature values (just tested with a cool-pack, give completely wrong values) (this works with my raspberry pi, python code)
also the example code does not wait 3 seconds with reading light value after writing...?

all in all the provided example code seems wrong to me... i reads values, but i think there is something wrong with the conversion of the values... would be nice if you could have a look into this  or give me some hints what i'm doing wrong... thx

Miceuz, 2015-11-13:
Hi,

I'm not exactly sure why you get weird readings. Are aware that light 
measurement takes some time, if you query the sensor before reading is 
done, it returns the old value. So basically you should wait for ~3 - 5 
seconds before reading light measurement result.

Thanks for the code!

Apollon77, 2015-11-14:
I think you need an "Unsigned int" for the light reading ... then you will never get negaive values.

For Temperature it could work with "signed" values ... will test the next days ...

Apollon77, 2015-11-15:
I tested the sensor today in "cold" environment an git interesting "light" values - it seems that the light sernsor don't like cold :-)
I've put the sensor in my -14 degree celsius fridge ... each reading in the following list is 3 seconds delay:
At start of the test the values wew at around 236-238 capaticitance, 22,7 degree celsius and ligth around 25/26.000 as you see in the first values. Then I moved to the fridge and light was changig, but this is also not interesting. Look at the timepoint where temperature in falling ...

Soil Moisture Capacitance: 238, Temperature: 22.70, Light: 25833
Soil Moisture Capacitance: 236, Temperature: 22.70, Light: 26611
Soil Moisture Capacitance: 237, Temperature: 22.70, Light: 26522
Soil Moisture Capacitance: 237, Temperature: 22.70, Light: 26289
Soil Moisture Capacitance: 238, Temperature: 22.70, Light: 26027
Soil Moisture Capacitance: 236, Temperature: 22.70, Light: 29537
Soil Moisture Capacitance: 237, Temperature: 22.70, Light: 33503
Soil Moisture Capacitance: 237, Temperature: 22.80, Light: 33672
Soil Moisture Capacitance: 237, Temperature: 22.80, Light: 31238
Soil Moisture Capacitance: 237, Temperature: 22.60, Light: 16495
Soil Moisture Capacitance: 237, Temperature: 22.70, Light: 16924
Soil Moisture Capacitance: 237, Temperature: 22.50, Light: 18798
Soil Moisture Capacitance: 238, Temperature: 22.70, Light: 44514
Soil Moisture Capacitance: 237, Temperature: 22.70, Light: 29258
Soil Moisture Capacitance: 237, Temperature: 22.80, Light: 49732
Soil Moisture Capacitance: 238, Temperature: 22.70, Light: 59615
Soil Moisture Capacitance: 239, Temperature: 21.20, Light: 60379
Soil Moisture Capacitance: 239, Temperature: 19.40, Light: 62946
Soil Moisture Capacitance: 238, Temperature: 18.00, Light: 65221
Soil Moisture Capacitance: 239, Temperature: 16.80, Light: 2377
Soil Moisture Capacitance: 240, Temperature: 15.90, Light: 5015
Soil Moisture Capacitance: 240, Temperature: 14.70, Light: 7588
Soil Moisture Capacitance: 241, Temperature: 13.70, Light: 10411
Soil Moisture Capacitance: 241, Temperature: 12.90, Light: 13260
Soil Moisture Capacitance: 241, Temperature: 11.90, Light: 15969
Soil Moisture Capacitance: 241, Temperature: 11.00, Light: 19006
Soil Moisture Capacitance: 241, Temperature: 10.30, Light: 21965
Soil Moisture Capacitance: 242, Temperature: 9.40, Light: 24788
Soil Moisture Capacitance: 243, Temperature: 8.70, Light: 27819
Soil Moisture Capacitance: 242, Temperature: 7.90, Light: 30899
Soil Moisture Capacitance: 242, Temperature: 7.20, Light: 33837
Soil Moisture Capacitance: 244, Temperature: 6.50, Light: 36629
Soil Moisture Capacitance: 244, Temperature: 5.80, Light: 39785
Soil Moisture Capacitance: 244, Temperature: 5.20, Light: 42828
Soil Moisture Capacitance: 245, Temperature: 4.60, Light: 45913
Soil Moisture Capacitance: 245, Temperature: 3.90, Light: 48760
Soil Moisture Capacitance: 245, Temperature: 3.40, Light: 51619
Soil Moisture Capacitance: 245, Temperature: 2.70, Light: 54787
Soil Moisture Capacitance: 246, Temperature: 2.20, Light: 57772
Soil Moisture Capacitance: 246, Temperature: 1.60, Light: 60723
Soil Moisture Capacitance: 246, Temperature: 1.20, Light: 63549
Soil Moisture Capacitance: 247, Temperature: 0.60, Light: 1006
Soil Moisture Capacitance: 247, Temperature: 0.20, Light: 3900
Soil Moisture Capacitance: 247, Temperature: -0.30, Light: 6955
Soil Moisture Capacitance: 247, Temperature: -0.80, Light: 9639
Soil Moisture Capacitance: 248, Temperature: -1.30, Light: 12380
Soil Moisture Capacitance: 248, Temperature: -1.80, Light: 15314
Soil Moisture Capacitance: 247, Temperature: -2.30, Light: 17992
Soil Moisture Capacitance: 248, Temperature: -2.80, Light: 20789
Soil Moisture Capacitance: 248, Temperature: -3.00, Light: 23535
Soil Moisture Capacitance: 248, Temperature: -3.50, Light: 25958
Soil Moisture Capacitance: 249, Temperature: -4.10, Light: 28836
Soil Moisture Capacitance: 249, Temperature: -4.20, Light: 31497
Soil Moisture Capacitance: 249, Temperature: -4.60, Light: 33924
Soil Moisture Capacitance: 249, Temperature: -5.20, Light: 36519
Soil Moisture Capacitance: 249, Temperature: -5.30, Light: 38986
Soil Moisture Capacitance: 249, Temperature: -5.60, Light: 41580
Soil Moisture Capacitance: 249, Temperature: -6.00, Light: 44005
Soil Moisture Capacitance: 249, Temperature: -6.20, Light: 46106
Soil Moisture Capacitance: 250, Temperature: -6.60, Light: 48673
Soil Moisture Capacitance: 251, Temperature: -6.90, Light: 50834
Soil Moisture Capacitance: 251, Temperature: -7.30, Light: 53472
Soil Moisture Capacitance: 251, Temperature: -7.50, Light: 55486
Soil Moisture Capacitance: 251, Temperature: -7.90, Light: 57828
Soil Moisture Capacitance: 251, Temperature: -8.00, Light: 59703
Soil Moisture Capacitance: 251, Temperature: -8.30, Light: 62157
[... 146 lines cut ...]
Soil Moisture Capacitance: 238, Temperature: 21.70, Light: 34436
Soil Moisture Capacitance: 237, Temperature: 21.70, Light: 34479
Soil Moisture Capacitance: 237, Temperature: 21.70, Light: 35180
Soil Moisture Capacitance: 238, Temperature: 21.80, Light: 37110
Soil Moisture Capacitance: 238, Temperature: 21.80, Light: 37160
Soil Moisture Capacitance: 237, Temperature: 21.80, Light: 37003
Soil Moisture Capacitance: 237, Temperature: 21.90, Light: 37267
Soil Moisture Capacitance: 238, Temperature: 21.90, Light: 37347
Soil Moisture Capacitance: 237, Temperature: 21.90, Light: 37342
Soil Moisture Capacitance: 238, Temperature: 22.00, Light: 37356
[... cut ...]
Soil Moisture Capacitance: 237, Temperature: 22.20, Light: 36745
Soil Moisture Capacitance: 237, Temperature: 22.30, Light: 36608
Soil Moisture Capacitance: 238, Temperature: 22.30, Light: 37242
Soil Moisture Capacitance: 237, Temperature: 22.30, Light: 37187
Soil Moisture Capacitance: 237, Temperature: 22.40, Light: 37129
Soil Moisture Capacitance: 237, Temperature: 22.40, Light: 36813

You can also see that after moving it our of the fridge the light values start normalizing again ...

Any idea?! Except that the sensor do't work well <0 degree celsius :-)

Miceuz, 2015-11-17:
Weel, yes, light measurement is a hack that worked ok for room temperature operated devices ;) If you apply large temperature swings to the sensor, light measurement goes crazy as it depends on AVR pin leakage current and on parasitic capacitance in the LED.

Apollon77, 2015-11-17:
:-)) ok for me personally because I will not really use it, but should maybe be noted on the info-page ...

sekdiy, 2016-06-03:
Shouldn't the light counter top out at 65534 (2^16 - 1) instead of 65535 (2^16) in main.c, line 129 (https://github.com/Miceuz/i2c-moisture-sensor/blob/master/src/main.c#L129)?

I'd recommend to `#include <limits.h> ` and then use the `UINT_MAX` constant.

I've noticed this possibility after experiencing a discontinuity in my readings:

[image]

krikk, 2016-06-04:
i have a few sensors in testing now and all i can say is, that the light reading is a bit jumpy if it comes to low light conditions ... have a look on my graphs for 5 i2c-moisture-sensors and one BH1750 light sensor:
[image]
...i first thought resetting the sensor before each reading will get better results, but this is not the case...  we all know the light reading is a "hack" so i can live with it...

Miceuz, 2019-03-03:
To get the idea how does the light sensing work, please look for "led as light sensor" posts on hackaday and whatnot. The basic idea being that parasitic capacitance is charged when reverse-biasing it. Then this capacitance discharges in two ways - via pin leakage and via photo electric effect - more light shines into the led, faster it discharges. By timing this discharge event one can get an estimate of light level there is. 

The method is very cheap, but very inaccurate - readings will depend wildly on ambient temperature and from sensor to sensor. Thus it is just suitable for rough estimate is it dark or not.

closed 2019-03-03 by Miceuz