Miceuz/i2c-moisture-sensor #3: Arduino Pull Up Resistors

SackLunch, 2015-12-08:
Do we need pull up resistors for an Arduino setup for pins A4 and A5 on the Arduino?  I have seen some comments on pages saying they are required but I'm not quite sure.

Any input appreciated.

Miceuz, 2015-12-08:
Yes, pullup resistors are to be provided by master.

SackLunch, 2015-12-09:
Thanks for the response.  I am having trouble running an I2C scanner and reading the Chirp!.  Works fine with other Arduino I2C products.  I've tried running 10k and 4.6k pull up resistors for Arduino A5 and A4 to 5 volts with similar results.  The scanner gets stuck and becomes non-responsive.  
When I try to read the capacitance the serial prints out 65531 when running your example scripts.
Am I missing something obvious with my wiring?

Connection 
Chirp pin 1 - no connection
Chirp pin 2 - Arduino VCC
Chirp pin 3 - Arduino A5 (with 4.6k jump up resistor)
Chirp pin 4 - Arduino A4 (with 4.6k jump up resistor)
Chirp pin 5 - no connection
Chirp pin 6 - Arduino GND

Miceuz, 2015-12-09:
What is your sensor version?

SackLunch, 2015-12-09:
I have version 2.4 with the reset button.

Miceuz, 2015-12-13:
Then you are using the wrong code. Use this: https://gist.github.com/Miceuz/8ace1cde27671e8e161d

closed 2017-04-17 by Miceuz

gluckgabor, 2017-09-12:
Hi! I am using sensor version 2.7.5., ruggedized, with white cable. I have tried three different code versions and four different pullup resistors: 270Ohm, 4,5kOhm, 7,5kOhm and 10kOhm. I have almost exhausted all the possible combinations of these. All of these consistently gave the upper bound of the measurement interval at all three sensors no matter if the sensor was in cup of water or in dry air (ok, the light sensor is coated in waterproof rubber hose, so it is not legit measurement anyway). I would love take out the guesswork from the setup.. Can someone please give me a hint about the size of the pullup resistor I must use? Thank You in advance.

Miceuz, 2017-09-12:
Hi, 

Probably your sensor is dead. Happens sometime. What was your order number? I will replace it for you. Also, can you send it back to me for analysis?

gluckgabor, 2017-09-12:
#79194, sure I will send it. Thank You!

gluckgabor, 2017-09-12:
Wrote to You on Tindie regarding your address. Thank You