Miceuz/i2c-moisture-sensor #4: compile error

pilotak, 2016-01-19:
Hi,
when i try to compile src with Makefile i get this error
```
avr-gcc-4.9.2 -DF_CPU=8000000    -Iusbdrv -I. -DDEBUG_LEVEL=0  -Os  -ffunction-s
ections -fdata-sections -save-temps  -Wl,--gc-sections,--relax -mmcu=attiny84
-c main.c -o main.o
main.c: In function 'adcSetup':
main.c:58:5: error: 'ADMUXB' undeclared (first use in this function)
     ADMUXB = 0;
     ^
main.c:58:5: note: each undeclared identifier is reported only once for each fun
ction it appears in
main.c: In function 'adcReadChannel':
main.c:71:5: error: 'ADMUXA' undeclared (first use in this function)
     ADMUXA = channel;
     ^
main.c: In function 'TIMER1_OVF_vect':
main.c:128:86: warning: 'TIMER1_OVF_vect' appears to be a misspelled signal hand
ler
 ISR(TIMER1_OVF_vect) {

      ^
main.c: In function 'wdt_disable':
main.c:169:5: error: 'CCP' undeclared (first use in this function)
     CCP = 0xD8;
     ^
main.c: In function 'wdt_enable':
main.c:174:5: error: 'CCP' undeclared (first use in this function)
     CCP = 0xD8;
     ^
main.c: In function 'setupPowerSaving':
main.c:238:79: error: 'PRTIM2' undeclared (first use in this function)
     PRR = _BV(PRTIM0) | _BV(PRTIM1) | _BV(PRTIM2) | _BV(PRSPI) | _BV(PRUSART0)
| _BV(PRUSART1); //shut down everything we don't use
                                                                               ^

main.c:238:97: error: 'PRSPI' undeclared (first use in this function)
     PRR = _BV(PRTIM0) | _BV(PRTIM1) | _BV(PRTIM2) | _BV(PRSPI) | _BV(PRUSART0)
| _BV(PRUSART1); //shut down everything we don't use

                 ^
main.c:238:114: error: 'PRUSART0' undeclared (first use in this function)
     PRR = _BV(PRTIM0) | _BV(PRTIM1) | _BV(PRTIM2) | _BV(PRSPI) | _BV(PRUSART0)
| _BV(PRUSART1); //shut down everything we don't use

                                  ^
main.c:238:134: error: 'PRUSART1' undeclared (first use in this function)
     PRR = _BV(PRTIM0) | _BV(PRTIM1) | _BV(PRTIM2) | _BV(PRSPI) | _BV(PRUSART0)
| _BV(PRUSART1); //shut down everything we don't use

                                                      ^
main.c:239:5: error: 'ACSR0A' undeclared (first use in this function)
     ACSR0A = _BV(ACD0); //disable comparators
     ^
main.c:239:21: error: 'ACD0' undeclared (first use in this function)
     ACSR0A = _BV(ACD0); //disable comparators
                     ^
main.c:240:5: error: 'ACSR1A' undeclared (first use in this function)
     ACSR1A = _BV(ACD1);
     ^
main.c:240:21: error: 'ACD1' undeclared (first use in this function)
     ACSR1A = _BV(ACD1);
                     ^
Makefile:55: recipe for target 'main.o' failed
make: *** [main.o] Error 1
```

what i'm doing wrong?

Miceuz, 2016-01-19:
You are compiling for attiny84 while the code is for attiny441. Attiny441 is not supported by avr-libc, avr-gcc and avrdude by default. You have to apply this patch before: https://github.com/adam-dej/avr-toolchain-patches

pilotak, 2016-01-19:
Oh, i over looked that, sorry. Is there a chance to run it on ATTINY84 or you specifically use those built-in analog comparators?

pilotak, 2016-01-23:
For Windows users like me, the easist way to install gcc and support for ATTiny441 is to install [WinAVR](http://sourceforge.net/projects/winavr/files/latest/download), download [Atmel AVR toolchain](http://www.atmel.com/tools/ATMELAVRTOOLCHAINFORWINDOWS.aspx) and copy all files from toolchain to the folder you installed WinAVR and you ready to rock

closed 2016-01-23 by pilotak