Miceuz/i2c-moisture-sensor #7: Attiny getCapacitance wrong?

breemts, 2016-07-07:
Hi,
I wanted to "update" my sensor and see if it makes a difference. Now the capacitance reading is 1023-1025. I tried to figure out why and thats what I noticed.
**getCapacitance** returns this: `1023 - (caph - capl)` and on my sensor capl is higher than caph. So I wanted to know why. With my DMM I measured pin 6(PA7) wich should be capl, pin 8(PA5) should be caph and pin 5(PA2) wich isn't mentioned in the code but is connected to the sensor-traces. 
pin6 _2.45V_; pin8 _2.46V_; and pin5 _2.43V_. 
When I manipulate the code that I get capl and caph on my console capLow is alwasy _1-2_ values higher than caphigh. capl values are 508-506 and caph values are _507-504_. 
So I thought my Sensor is broken, but what part? I measured all **components** and they are all **fine**. 

Than I took another sensor that I have and measured the Voltage on the pins.
pin6 _4.88V_; pin8 _4.94V_; pin5 _4.85V_
I get readings about 230 from this sensor.

So could it be that the code for the Attiny is wrong? When I do the calculation with the formal in the code and use the values of the working sensor I get this: `1023 -(991[4.94V] - 979[4.88V]) = 1011`
So the working Sensor must have **different code**. 
And I think the Voltage difference is because **PA2 the "Exitation" pin is not mentioned in the code**. I think it should be set to "HIGH" or _1024_, something like this.

Would be really if you could update your code. And what kind of **IDE** did you use for programming the Attiny? I'm uploading over a Arduino via ISP.

Best.
Benjamin Reemts

madmax76, 2016-11-07:
> And I think the Voltage difference is because PA2 the "Exitation" pin is not mentioned in the code. I think it should be set to "HIGH" or 1024, something like this.

Excitation is done by setting the CKOUT Fuse, see the Makefile. Did you check that? If I am not mistaken, the full 16MHz is put onto the sensor track via the 330Ohm resistor. I am currently investigating if it's possible to run without the external clock source (16MHz Crystal). 

I have used both Atmel Studio as well as USBAsp with AVRdude to program the device.

Miceuz, 2016-11-07:
I can guarantee that the code on the github is the same one I use to program sensors. It is possible to run on internal RC oscillator, but the main reason of adding a crystal was improved stability - internal oscillator drifts a lot with ambient temperature.

closed 2017-04-17 by Miceuz