Miceuz/i2c-moisture-sensor #8: Suddenly weird reading?

solars, 2016-07-16:
Hi there,

I'm using the sensor with a Wemos D1 Mini, it worked nicely. After changing some code (not related to the sensor), I suddenly get readings:

Temperature: -0.10 (float temp = sensor.getTemperature()/(float)10;)
Light: 4294967295
Firmware version: FF
Sensor Address: 20

Can anyone tell me what went wrong?

Thank you,
Christoph

montreuilg, 2016-11-04:
I have the same problem, using the example code provided here!

I2C Soil Moisture Sensor Address: 20
Sensor Firmware version: FF
Soil Moisture Capacitance: 65535, Temperature: -0.10, Light: 65535
---------------------------------------------

solars, 2016-11-05:
Check if you are using the right pins, I think in my case I swapped them, but don't remember exactly..

closed 2016-11-05 by solars

gluckgabor, 2017-11-01:
Hi All, 

I have the same problem. Can't figure out what is the issue. There are no special pins to be referenced in the code..  Care was taken to observe these 4, and pullup resistors are in place: RED - VCC, BLACK - GND, GREEN- SDA, YELLOW - SCL, it was working, then it suddenly wrong readings exactly same as the above reported ones..

mindforger, 2017-11-01:
maybe there is a chance to break the i2c on the attiny in some way by using odd timing (talk about edge cases) you can try to do a reset from your host controller by toggeling the SCL manually (open drain, so only set the output to low and toggle the direction of the pin) for a few times (at least 8, to clear off the shift register) ... or maybe toggle the power line through eitehr direct a port pin or using an transistor

BEWARE! always turn off the I2C master and set the SCL and SDA pins to low and output to prevent crossfeeding any power BEFORE turning off the VCC of the sensor to reset it!

gluckgabor, 2017-11-01:
Hi Mindforger!

Thank You for superfast reply.

I understand your instructions to be this steps:

1. Set the SCL and SDA pins to "low and output" and toggle the direction of these pins for at least 8 times to clear off the shift register.
2. Or toggle the VCC power line through both directions via a port pin.
3. Turn off the I2C master.
4. Turn off the VCC of the sensor to reset it.

Question now is how to still power the sensor after turning off the I2C master.. as the I2C master is supplying the sensor with VCC... I understand your warning so that I have to turn off the I2C master first and only after that I can turn off the sensor..

For the toggling I have written the following, is this what You meant? (My microcontroller type is Genuino Zero btw...)

pinMode(SCL, OUTPUT);
pinMode(SDA, OUTPUT);
digitalWrite(SCL, HIGH);   // sets  on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
digitalWrite(SCL, HIGH);   // sets on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
digitalWrite(SCL, HIGH);   // sets  on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
digitalWrite(SCL, HIGH);   // sets on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
digitalWrite(SCL, HIGH);   // sets  on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
digitalWrite(SCL, HIGH);   // sets  on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
digitalWrite(SCL, HIGH);   // sets on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
digitalWrite(SCL, HIGH);   // sets  on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
digitalWrite(SCL, HIGH);   // sets on
digitalWrite(SDA, HIGH);   // sets on
delay(200);                  // waits 
digitalWrite(SCL, LOW); 
digitalWrite(SDA, LOW);
delay(200); 
pinMode(SCL, INPUT);
pinMode(SDA, INPUT);

Thank You in advance!

gluckgabor, 2017-11-01:
I have discovered that the erroneous data is displayed at the serial monitor even when the sensor is not connected... puzzles me more and more..

mindforger, 2017-11-01:
thats kind of normal, I2C is a Pull down Bus, it has a continous high level if not touched, all slaves and master only pull down, so there can never be a short circuit, if you are reading 0xFFFF this means, the slave did not respond to your read request and when it does not touch the line, that means you get 16 1-bits reading 0xFFFF or 65535 ... your issue is, that the slave maybe responds to your address byte (check if you get an acknowledge on the address byte) but does not shift out any data for whatever reason ... try to reduce the speed for testing, the sensor can adapt to all known I2C speeds as far aws i know

gluckgabor, 2017-11-01:
Seems like I2C address scanning solved the problem, found on this link: [http://playground.arduino.cc/Main/I2cScanner](http://playground.arduino.cc/Main/I2cScanner) However I am still puzzled how the I2C address of this slave have changed seemingly all by itself.. :)

Miceuz, 2017-11-01:
@gluckgabor Did you by any chance disconnect the sensor while the bus was running? Maybe connection was flaky? Sensors have this drawback - I did not add any protection on set address command and a stray communication on the bus can be interpreted as an address set command.

gluckgabor, 2017-11-01:
Yes, that is fully possible. Thanks for explanation! Have a nice day.