Miceuz/i2c-moisture-sensor #11: Sensor outputs only 65535 as the reading

varunhr, 2017-04-18:
Hi i recently bought a chirp I2C soil moisture sensor V2.7.4 and i hooked it up to Arduino Uno Like this
Uno Chirp
5V/3.3V----- VCC(Tried Both)
Gnd---- GND
SDA -----SDA(10k Pullup)
SCL------ SCL(10k Pullup)

My Code is this
-----------------------------------------

#include <Wire.h>

void writeI2CRegister8bit(int addr, int value) {
Wire.beginTransmission(addr);
Wire.write(value);
Wire.endTransmission();
}

unsigned int readI2CRegister16bit(int addr, int reg) {
Wire.beginTransmission(addr);
Wire.write(reg);
Wire.endTransmission();
delay(20);
Wire.requestFrom(addr, 2);
unsigned int t = Wire.read() << 8;
t = t | Wire.read();
return t;
}

void setup() {
Wire.begin();
Serial.begin(9600);
writeI2CRegister8bit(0x20, 6); //reset
}

void loop()
{
Serial.print(readI2CRegister16bit(0x20, 0)); //read capacitance register
Serial.print(", ");
Serial.print(readI2CRegister16bit(0x20, 5)); //temperature register
Serial.print(", ");
writeI2CRegister8bit(0x20, 3); //request light measurement 
Serial.println(readI2CRegister16bit(0x20, 4)); //read light register
delay(1000);
}

-----------------------------------------------------
My Serial port output is this

--------------------------------
65535, 65535, 65535
65535, 65535, 65535
65535, 65535, 65535
65535, 65535, 65535
65535, 65535, 65535
65535, 65535, 65535
--------------------------------

Miceuz, 2017-04-18:
Hi,

How long if your cable? Also, please write directly to me - mic at wemakethings net. Please send me your order number, we will sort it out.

closed 2017-05-07 by Miceuz

varunhr, 2017-05-08:
Hi,
    Sorry for the delay in response. The sensor worked properly after some
solder touchup on the Crystal Oscillator. Thanks for your support.

Regards
Varun h r