Miceuz/i2c-moisture-sensor #14: Soil Moisture Capacitance depends on power supply voltage

bilbolodz, 2017-08-02:
Value returned by getCapacitance() depends on power supply voltage. In the same physical conditions (power supply "about" 3,3V) dry sensor:

- voltage 3,44 V ADC value 346
- voltage 3.28 V ADC value 366

You are using Vcc as analog reference (  ADMUXB = 0 in adcSetup() function). It's probably better to use "Internal 1.1V or 2.2 V" as reference for ADC

Miceuz, 2017-08-16:
Yes, it is true. You have to provide a stable voltage for the sensor. I have solved this in my [rs485](https://github.com/Miceuz/rs485-moist-sensor) sensor version by adding an on-board regulator, maybe I will pull this change to i2c sensor sometime.

closed 2017-08-16 by Miceuz

bilbolodz, 2017-08-16:
My soil moisture sensors are battery powered. Adding on-board power regulator to such sensors it's just a waste of power and it's not a good idea. It should be corrected in software.

Miceuz, 2017-08-16:
Then you can provide regulation for all of the sensors at once. You are welcome to share the calibration table if you go the software route.

bilbolodz, 2017-08-16:
I'm using you sensor with "own build" Mysensors node with NFR24L01 transport as "main CPU". There are battery powered  and power regulator is not needed because there are powered directly from CR123 battery. Arduino and NRF24L01 are working good "down to" 2.8V voltage (actually there are working even at 1.8V if atmel is using internal 1MHz oscillator). I think you sensor also should work at 2.8V so that's why I'm asking to use internal 1.1V reference instead of external power supply.

Miceuz, 2017-08-16:
Yes, I could use internal 1.1V reference, but this creates some issues too - I need to generate the test square wave at 1.1V peak to peak then, this means, I have to create 1.1V source on board, add level shifting, buffering, etc - this adds to the price and my aim is to provide a cheap sensor.  I could use internal 2.2V reference and no level shifting, but then again, some ADC readings will be over the range when the battery is full and then I'm limited to 3V power only. Tradeoffs have to be made somewhere and I am making one which impacts the price the most. 

BTW idea about using software compensation is really interesting and might work. Please share if you get any good results.

Lastly - this is an open source project, you can download the firmware and do whatever with it, also you can modify hardware.

In my next batch of sensors I have made a change that will really improve the battery life - I will be powering down thermistor divider circuit that now hogs the major part of power consumption in sleep - 150uA

bilbolodz, 2017-08-16:
OK, thank your for your statement. I will think if I'm able to deal with these issue in software way without modifying current hardware.

bilbolodz, 2017-08-19:
I think it could be good idea to add command to measure "system voltage". I requires to stop square wave give "0" measure SENSE_LOW, give "1" measure SENSE_HIGH and report it. That should allow to compute correction factor by main CPU (or even it could be corrected in sensors itself). No hardware modification is needed it also doesn't break backward compatibility with previous code.

Miceuz, 2017-08-23:
It is not how it works with ADC - you will get 0 and 1023 as readings because adc reference is system voltage. If I use internal reference, then system voltage is way above the reference, thus I need a hardware change - resistor divider to measure system voltage.

bilbolodz, 2017-08-23:
I know how ADC works. There is already 1:1 divider on the board (two 1M resistors). In case of 3V power supply it's enough to measure power voltage using 2.2V internal reference, in case of 5V it's problematic....The other issue is: is it usable information and is it possible to compute correction from it. I thought about it but got no solution so far.

commit f0a01a20 (2019-03-03): Update README.md

Close #14 fix broken link