Miceuz/i2c-moisture-sensor #16: Invalid data after some time

lobeck, 2017-08-12:
I've setup the sensor with a Adafruit Feather HUZZAH (ESP8266) and can read data for a while (1-3 days) then suddenly i only get 65k capacitance and invalid temperatures. The connections are soldered on a prototype board and 10k pull-ups

Power supply is a Anker wall adapter, so i don't think it's an issue.

Here's the relevant code (stripped the network pieces):

```
#include <ESP8266WiFi.h>
#include <WiFiUDP.h>
#include <I2CSoilMoistureSensor.h>
#include <Wire.h>

int status = WL_IDLE_STATUS;     // the Wifi radio's status
WiFiUDP udp;

I2CSoilMoistureSensor sensor;

void setup() {
  Wire.begin();
  Serial.begin(9600);

  Serial.println("Attempting to connect to WPA network...");
  udp.begin(4097);

  sensor.begin(); // reset sensor
  delay(1000); // give some time to boot up
  Serial.print("I2C Soil Moisture Sensor Address: ");
  Serial.println(sensor.getAddress(),HEX);
  Serial.print("Sensor Firmware version: ");
  Serial.println(sensor.getVersion(),HEX);
  Serial.println();
}

void loop() {
  while (sensor.isBusy()) delay(1000); // available since FW 2.3

  int capacitance = sensor.getCapacitance();
  float temperature = sensor.getTemperature()/(float)10;

  sensor.sleep(); // available since FW 2.3
  udp.beginPacket({192, 168, 2, 48}, 8089);

  Serial.print("Soil Moisture Capacitance: ");
  Serial.print(capacitance); //read capacitance register
  Serial.print(", Temperature: ");
  Serial.println(temperature); //temperature register

  udp.print(chirpLine);
  udp.endPacket();

  delay(10 * 1000);
}
```

Any ideas what might cause this? Currently testing a version without the `sensor.sleep()`, because i haven't found any statements regarding wakeup.

Miceuz, 2017-08-16:
Hi,

Probably it's a combination of long cable and esp8266 side treating clock stretching poorly. Have a look in this thread where we were debugging something similar to this:
https://github.com/Apollon77/I2CSoilMoistureSensor/issues/8

lobeck, 2017-08-16:
Haven't seen Apollon77/I2CSoilMoistureSensor#8 as relevant in my case, as i only got the issue after some time. Got the rugged version too, so cable length wouldn't be my fault ;) It's currently mitigated by not using `sensor.sleep()`, but maybe i'll move away from the esp8266 too and move to some other low power solution. Without the sleep it's running now for 5 days and i'll further look into this once it's happening again.

closed 2017-08-16 by lobeck