Miceuz/i2c-moisture-sensor #19: Gerbers for 2.7.5

bilbolodz, 2017-09-01:
Could you please provide gerber files for 2.7.5 version? It looks that at github there are for 2.7.1. I've tried to generate it myself but kicad gives some error:

`Date of KiCad version required (or newer): 01/23/17

Full error text:
PARSE_ERROR: Expecting 'clearance, trace_width, via_dia, via_drill, uvia_dia, uvia_drill, or add_net' in input/source
'/home/bilbo/robota/i2c-moisture-sensor/i2c-moist-sensor-pannel.kicad_pcb'
line 132
offset 6
from dsnlexer.cpp : Expecting() : line 369`

when I'm trying to open i2c-moist-sensor-pannel.kicad_pcb file

2ni, 2018-07-24:
Just clone the whole repo somewhere to your machine, then do "File/Open Project" on kicad and navigate to where you cloned the repo.
Open "i2c-moist-sensor-pannel.pro". You will then see a list of files in kicad itself, and you can double-click on i2c-moist-sensor.kicad_pcb

That worked for me.

Miceuz, 2018-07-24:
You have to use KiCad 5, it has been released last week.

closed 2019-03-03 by Miceuz