Miceuz/i2c-moisture-sensor #20: Measuring capacitance: Frequency counting?

DFyson, 2017-12-05:
I noticed you mentioned you had tried timing the period of the sensor in an RC oscillator but didn't work so well because the capacitance was so low thus requiring a combination of very high resistance/fast timer. Have you considered counting the frequency instead? Make an RC oscillator (such as SN74LVC1G132 since it can be disabled) then use the 16bit counter on the microcontroller as a frequency divider and feed that into an interrupt. The linearity, dynamic range and resolution should be much better than the current low-pass filter/ADC method and the circuit would be more versatile for other frequency counting or capacitance applications.

Miceuz, 2017-12-05:
Yes, counting frequency is another way to measure the capacitance. I did not try to do this, but gate-based oscillators are pretty quirky from my experience, so I am afraid it would be not that stable especially over the temperature and over multiple units. Building a precision oscillator would be a non trivial and costly task. Will have to try this some time.

closed 2017-12-05 by Miceuz

mindforger, 2017-12-05:
>especially over the temperature

i can support this, had oscillator based solutions drifting like hell, this one works fairly stable even outside