Miceuz/i2c-moisture-sensor #27: Capacitive measurement

rubengr, 2018-10-30:
Hi,
I bought one chirp, and make another to test diferences.. I read from https://wemakethings.net/chirp/ that "chirp uses capacitive sensing to measure the moisture. 1MHz square wave (system clock is used for that) is output from the chip through a resistor into a big pad that together with the surrounding ground plane forms a parasitic capacitor. "

What i'm see in this code is just a diference between two readings. Can you explain how do you read the capacitance? 

`uint16_t getCapacitance() {
    capMeasurementInProgress = 1;
    uint16_t caph = adcReadChannel(CHANNEL_CAPACITANCE_HIGH);
    uint16_t capl = adcReadChannel(CHANNEL_CAPACITANCE_LOW);
    capMeasurementInProgress = 0;
    return 1023 - (caph - capl);
}`

Thanks

Miceuz, 2018-10-31:
I am reading the capacitance the same way as described, but the idea is extended a bit further. I am measuring both peak and dip levels of the resulting triangle wave, capacitance estimate is a difference of those two numbers. Note, I don't return a true capacitance value in Farrads, just an arbitrary number that changes with changing capacitance.

closed 2018-10-31 by Miceuz

rubengr, 2018-10-31:
How do you know in microcontroller if we are on peak or dip level?