Miceuz/i2c-moisture-sensor #29: Question about the excitation pin

Eriobis, 2019-10-08:
Hi,

I am doing a bluetooth version of your project so I have used your schematics and I was wondering what is exactly the use of the excitation pin ? It seems like I don't see any call to this pin in the code. Can you help me with that ?

Miceuz, 2019-10-08:
Hi there, I have also done a BLE version here - https://github.com/Miceuz/ble-soil-moisture-sensor
Excitation pin is used to excite the sensor. It simply outputs square wave. It is used to output main clock frequency via special registers on AVR.

Eriobis, 2019-10-08:
Wow thanks for your prompt answer. I know I have seen it somewhere in the past, I have designed my board with this pin on a PWM output, but could not find the freq. and duty cycle.

I haven't seen your BLE version of it ! Is it on tindie as well ?

closed 2019-12-25 by Miceuz