Miceuz/i2c-moisture-sensor #30: Qwiic connectors

la7dja, 2019-12-24:
Hi

I think it would would be great to have [Qwiic connectors](https://www.sparkfun.com/qwiic) as an option for indoor use. Since it was space on the PCB to add this with only minor changes to the layout, I did that in a [fork](https://github.com/la7dja/i2c-moisture-sensor). [This](https://github.com/la7dja/i2c-moisture-sensor/blob/master/i2c-moist-sensor-qwiic.png) is how looks (without the Qwiic logo which I added later). If it is of interest I could make an pull request, but I don't know how well that will work with Kicad files.

Miceuz, 2019-12-24:
Hi there, good idea! Send me the pull request, I will see how it goes. Thanks!

olNick, 2020-07-01:
Hi,
This is the only way I could find to contact you...
Attached is a procedure we used to get a transfer function for the i2c sensor

[M&M_Soil Moisture Sensor Calibration.docx](https://github.com/Miceuz/i2c-moisture-sensor/files/4857624/M.M_Soil.Moisture.Sensor.Calibration.docx)

Hope it helps someone.
Soil used was primarily clay....

regards,
nick

olNick, 2020-07-01:
Sorry,
To save you some reading....

Example:
The calculated dry basis moisture content (MCd) of the soil when a soil moisture sensor reads a capacitance value of 550 corresponds to:
MCd = (0.000001007*5502)+(0.000024885*550)-0.062508722
MCd = 0.2558 or 25.58 %

regards,

Miceuz, 2020-07-18:
Hello,

Thanks a lot for this! Really appreciated!

Can I share it with community? The doc is missing an author/date I would like to attribute you.

olNick, 2020-07-20:
> Links:

Sure, that's why I sent it...

 As authors,
 Nikos Boretos, George Kazakis, Dany Ghosn, Mediterranean Agronomic 
 Institute of Chania (MAICh)

 date 02/2020

 regards,
 Nikos

Miceuz, 2020-07-20:
Great, thanks!

One more question - were sensors powered from 5V or 3V3?

olNick, 2020-07-20:
3v3