Miceuz/i2c-moisture-sensor #32: Cannot change the address with RasPi/Python script.

DIY-Blub, 2020-05-07:
**This entry is primarily for documentation purposes, should the problem occur with others, please write.**

Soil moisture sensor v2.7.8 / version 0x26 (purchase date: April 2020)
Raspberry Pi 1 Model B, Rev 2; "Buster" with latest updates

I wanted to change the I2C address of my 3 purchased sensors with the Raspberry Pi (Python). The sensor query (read out sensor values) per device at 0x20 works. Changing the address was done without error message, but the sensor kept the default address 0x20.

**Python2: used code from**
**https://github.com/ageir/chirp-rpi**
```
wget -O Chirp_ageir.py https://raw.githubusercontent.com/ageir/chirp-rpi/master/chirp.py
chmod +x Chirp_ageir.py
python2 ./Chirp_ageir.py 0x20 set 0x24
```
> Chirp I2C address changed to 0x24

After one (and more) restarts the I2C address is still 0x20.

**https://github.com/JasperWallace/chirp-graphite**
```
wget -O Chirp_JasperWallace.py https://raw.githubusercontent.com/JasperWallace/chirp-graphite/master/chirp.py
chmod +x Chirp_JasperWallace.py
python2 ./Chirp_JasperWallace.py 0x20 --set-address 0x24
```
> version 0x26
> Setting the chirp's i2c address to 0x24
> done

After one (and more) restarts the I2C address is still 0x20.

**https://github.com/Miceuz/i2c-moisture-sensor#raspberry-pi-examples**
Also this simplified script (by JasperWallace) did not change the address, after some adjustments.

**Solution with Arduino**
Only with an Arduino Uno I was able to change the I2C addresses. I used this library:
https://github.com/Apollon77/I2CSoilMoistureSensor

Miceuz, 2020-05-07:
Hi there, thanks for pointing this out. I have updated raspi code on github readme page:

`	def set_addr(self, new_addr):
		# To change the I2C address of the sensor, write a new address
		# (one byte [1..127]) to register 1; the new address will take effect after reset
		self.bus.write_byte_data(self.address, 1, new_addr)
		# second request is required since FW 0x26 to protect agains spurious address changes
		self.bus.write_byte_data(self.address, 1, new_addr)
		self.reset()
		self.address = new_addr
`

DIY-Blub, 2020-05-07:
ok, thanks!

closed 2020-05-07 by DIY-Blub