Miceuz/i2c-moisture-sensor #35: Temperature Funky above 25°

vrenaville, 2021-06-28:
Hello,

I currently have a rugged version of i2c. It's connected to a RAK 5205.

If Temperature raise above 25° temporature goes funky, 0 c° , 12°c etc ....
when the night comes and the temperature goes below 25c°, it's fine.

My setup:
- RAK 5205 (STM32L Lorawan device)
- Dupont cable connecte (male - male) of 20 cm
- Rugged version connected to 3.3v (on battery)

Do you have this kind of issue in the past ?

Kind regards,

Vincent

mindforger, 2021-06-28:
i noticed that too, on hotter days it seems the value is jumping heavily

vrenaville, 2021-06-28:
I was related to my code I only read the first byte :D

closed 2021-06-28 by vrenaville

Miceuz, 2021-06-28:
> i noticed that too, on hotter days it seems the value is jumping heavily

Please check if you are reading two bytes from the sensor. Also if sun shines onto the sensor this will make the temperature fluctuate with clouds, shadows, etc covering the sun.