Miceuz/i2c-moisture-sensor #36: Sleep mode: Rak module require datas

vrenaville, 2021-06-28:
Hello,

With RAK811 module, It do not support no data writing on register.
Do you know if I can write data on the register to set him to sleep ?

Do you know if I can write a byte on the register ?

Kind regards,

Vincent

Miceuz, 2021-06-28:
Hi @vrenaville what standard are they refering to? I am not aware of this limitation. 

No way around it from what I have seen.

One hack you can try is write 0x08 into register 0x08. If I am right, the sensor will go to sleep and come back from it immediately after receving 0x08 and will interpret that as sleep command and go to sleep again.

vrenaville, 2021-06-29:
I have tried to write 0x08 byte on register 0x08, it's accepted, but after the I2C bus and de sensor is completely stuck. 
Need to reset the system.

Anyway it's more a Rak issue than from your sensor, but let's wait they improve their tools

closed 2023-02-25 by Miceuz