Miceuz/i2c-moisture-sensor #37: ESP32/ MQTT/ LoRa version?

samuk, 2021-11-05:
Hi I realize this is not trivial.

It occurred to me that if you did a version with a ESP32

It could run https://docs.openmqttgateway.com/

And use LoRa mesh (not LoRaWAN) to connect back to the internet: https://meshtastic.discourse.group/t/meshtastic-to-build-environmental-networks/2628/26

Miceuz, 2021-11-10:
Hi @samuk, thanks for suggestions, I will look into that. Note that I have released a LoRaWAN board designed to work with my sensors: https://www.tindie.com/products/miceuz/low-power-lora-i2c-rs485-node/

Miceuz, 2023-02-25:
Also there is an ESP32 based board to hook up the sensor to WiFi and an e-paper display: [https://github.com/Miceuz/paper-wifi/](https://github.com/Miceuz/paper-wifi/)