Miceuz/i2c-moisture-sensor #38: No response to I2Cdriver

krisradio, 2022-04-26:
Hello,
I am using the moisture sensor for a smart greenhouse project. To familiarize myself with the I2C interface, I am first attempting to get it working with the [I2CDriver](https://github.com/jamesbowman/i2cdriver) product. Unfortunately, I am not getting any meaningful response; no matter what register I read from, the response is always 0xFF. I have not attempted a write, since it doesn't seem to be working right and I don't want to screw it up.

Here is an example:

```
C:\Users\7hesa>i2ccl COM10 w 0x00 2 r 0x20 2
0xff,0xff

C:\Users\7hesa>i2ccl COM10 w 0x02 2 r 0x20 2
0xff,0xff

C:\Users\7hesa>i2ccl COM10 w 0x04 2 r 0x20 2
0xff,0xff

C:\Users\7hesa>i2ccl COM10 w 0x05 2 r 0x20 2
0xff,0xff

C:\Users\7hesa>i2ccl COM10 w 0x07 2 r 0x20 2
0xff,0xff

C:\Users\7hesa>i2ccl COM10 w 0x09 2 r 0x20 2
0xff,0xff
```

The complete [I2CDriver documentation is available here](https://i2cdriver.com/i2cdriver.pdf). I will try the moisture sensor next with my microcontroller; if that _does_ work, I will bring this issue to the I2CDriver manufacturer. Thanks for reading!

System info: Windows 10 Home 10.0.19043, Build 19043, 64-bit

Miceuz, 2022-04-27:
I don't know where exactly my I2CDriver is, but I think you issue the command wrong, try:
`i2ccl COM10 w 0x20 0x00 r 0x20 2`

First argument in `w` command is the device address, 0x20 being the default address of the sensor. Let me know if it helps.

kriskeillor, 2022-05-08:
@Miceuz, thanks for getting back to me! I just saw this.
I messaged you on Tindie yesterday, feel free to ignore that since we're talking here.

I appreciate the help, its definitely gotten me further, although I'm still a bit confused.

That command seems to be working:

```
C:\WINDOWS\system32>i2ccl COM10 w 0x20 0x00 r 0x20 2
0x00,0xf9

C:\WINDOWS\system32>i2ccl COM10 w 0x20 0x00 r 0x20 2
0x00,0xf7

C:\WINDOWS\system32>i2ccl COM10 w 0x20 0x00 r 0x20 2
0x00,0xf8

/* Now I put my hand around the sensor... */

C:\WINDOWS\system32>i2ccl COM10 w 0x20 0x00 r 0x20 2
0x02,0x26

C:\WINDOWS\system32>i2ccl COM10 w 0x20 0x00 r 0x20 2
0x02,0x23
```

I'm getting variable but consistent (and non-0xFF) readings for the capacitance.

However, I still feel like something is wrong, because when I query the sensor's address, it reports 0xFF - which is obviously not true, since I'm speaking with address 0x20. First I reset and then ask for the address in one line:

```
C:\WINDOWS\system32>i2ccl COM10 w 0x20 0x06

C:\WINDOWS\system32>i2ccl COM10 w 0x20 0x02 r 0x20 1
0xff
```

I also can't seem to _change_ the address.

If it helps, the device version is apparently 0x26:

```
C:\WINDOWS\system32>i2ccl COM10 w 0x20 0x07 r 0x20 1
0x26
```

Thanks again for looking at this problem, I'm new to using i2cdriver so I really appreciate the pointers and the insight about your product. 

One more questsion: The [i2cdriver documentation](https://i2cdriver.com/i2cdriver.pdf) has register addresses for writes written in decimal, but you are writing register addresses in hex; do you know if one way is preferred?

Regards,
Kris

Miceuz, 2022-05-08:
Hi Kris,

To change the address you have to perform the register two times in a row, that's my workaround to fight against noise on lines changing address to random number - happened on several occasions.

Regarding the address read - are you allowing enough time for sensor to boot (about 1 second) 

I suggest having a look at the sensor library for Arduino environment: https://github.com/Apollon77/I2CSoilMoistureSensor You can the implementation for all the functions there.

kriskeillor, 2022-05-08:
Oh, I didn't see that you have to perform that write twice in a row. Is that documented anywhere? 
But, trying it that way did not work either. 

I did allow enough time for the boot, yes.

I forgot to mention one thing - the current draw jumps from 5 mA to 10 mA briefly when I send the restart, so I think that command is going through.

Thank you for the library link! I am going to switch from i2ccl to the i2cdriver python module, as its much better documented, to troubleshoot. My project is in C (or Cpp, not currently though), so I appreciate having that library to reference. For the time being python might be quicker to just get something working.

Will report back shortly..!

kriskeillor, 2022-05-08:
I'm having more success with the Python module and libraries and will probably stick to that over i2ccl. It's much better documented both from the official sources, and with open source contributions.

Thanks again for your help @Miceuz!

closed 2022-05-08 by kriskeillor