Miceuz/i2c-moisture-sensor #39: Mosfet instead of Q2 and Q3 bipolar ?

pit34, 2022-05-07:
Hello,

I ask myself why does Q2 and Q3 are bipolar tranistor and not mosfet ?
There is an explanation of operation here :
https://wemakethings.net/2012/09/26/capacitance_measurement/
But not why bipolar instead of mosfet to discharge the capacitor.

Thanks

kriskeillor, 2022-05-08:
Just a guess from me, but BJTs are cheaper and they do the same thing. That could be the reason.

Miceuz, 2022-05-08:
Err, I don't remember exactly, but I think BJTs I had in my bin were faster to switch than mosfets I had. But in general I had to scrap the idea of discharging as it creates a lot of complication (switching transistor really fast is not simple) and does not give a lot of pros.

closed 2022-05-08 by Miceuz