Miceuz/i2c-moisture-sensor #41: I2C all of a sudden stops working

sebastianneubauer, 2023-02-23:
I bought two i2c-moisture sensors a long time ago in 2015 with version v2.7.1. I never really tried them but now wanted to use them using micropython and an ESP32. It worked like a charm in the beginning, but then all of a sudden the first one stopped working, the micropython exception says:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/chirp.py", line 49, in reset
  File "/lib/chirp.py", line 44, in _register_8
OSError: [Errno 19] ENODEV
```
which means: there is not device at this address. Also a i2c scan doesn't return anything back. 
So i tried the second old sensor, but also there, after some hours, all of sudden, same problem, sensor seems to become dead. 

My setup:
VCC is 3.3V, 2k2 pullup resistors on both SCL and SDA

The micropython code is:
```
>>> from machine import Pin, SoftI2C
>>> import chirp
>>> i2c = SoftI2C(scl=Pin(22), sda=Pin(21))
>>> chirp_sensor = chirp.Chirp(i2c)
>>> chirp_sensor.reset()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/lib/chirp.py", line 49, in reset
  File "/lib/chirp.py", line 44, in _register_8
OSError: [Errno 19] ENODEV
```

What I tried:
* I toggeled SCL low and high several times, same with SDA, no effect
* I pulled "RESET" line low and high several times, the led flashes when pulled high, but no effect

Is there any way to bring them back, or are they dead? Any idea what could have killed them both?

Thanks a lot!

Miceuz, 2023-02-25:
Hi,

2015 was long time ago...

```
commit ba48c0ac055df5c0bccb623ff6fc83b1be8b0bf2
Author: miceuz <mic@wemakethings.net>
Date:   Wed Feb 17 00:17:30 2016 +0200

    Enable BOD finally!
```

Most probably flash is being corrupted due brown out. You should re-flash the sensor with the latest firmware.

closed 2023-02-25 by Miceuz

sebastianneubauer, 2023-02-25:
Sure, that is a long time ago, but why throwing them away if they are still good (if the are...). I am totally happy to buy a new rugged one, just I somehow would want to be sure that the new one doesn't get bricked again after few hours of runtime because I am doing something wrong.
Any hints or documentation how to flash new firmware?

Miceuz, 2023-02-25:
Sure, no need to throw them away. It's a standard AVR flashing. Are you familiar with that? Are your sensors rugged or plain?

sebastianneubauer, 2023-02-25:
They are both plain, a reason I would anyhow buy a rugged one. Sadly no clue about AVR flashing. Would be awesome if you could throw some pointers my way, e.g. do I need special hardware, where would I connect the sensor, which firmware to use, etc..

Miceuz, 2023-02-26:
Good that you have plain sensors, this means you have access to the programming port.
Here is the pinout for the programming header:
[image]

It's Sunday and I don't have hardware in my hands to double-check it myself, but these links can help:

This one probably is the most relevant for you:
https://learn.adafruit.com/stand-alone-programming-avrs-using-circuitpython

These are C++ based:
https://github.com/lbernstone/ESP_AVRISP
https://github.com/ESP32-Musings/OTA_update_AVR_using_ESP32

In order for you not to compile the sensor firmware from scratch, you can down download it here: https://gist.github.com/Miceuz/9762f93cad23a95064086fe6c73c575a

I will look into this more next week if you still need it, let me know how it goes.

Miceuz, 2023-02-26:
I have updated the Adafruit example so it has correct chip description to the one used in the sensor, see it below. It "should just work", but as I've said, I can't test it today.

``` python
# SPDX-FileCopyrightText: 2021 ladyada for Adafruit Industries
# SPDX-License-Identifier: MIT

"""
ATtiny441 programming example, be sure you have the '13a wired up so:
  ATtiny441 GND to CircuitPython GND
  ATtiny441 VCC to CircuitPython USB
  Pin 2 -> CircuitPython SCK
  Pin 1 -> CircuitPython MISO
  Pin 0 -> CircuitPython MOSI
  RESET  -> CircuitPython D5 (or change the init() below to change it!)
Drag "main.hex" onto the CircuitPython disk drive, then open REPL!
"""

import board
import busio
import adafruit_avrprog

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
avrprog = adafruit_avrprog.AVRprog()
avrprog.init(spi, board.D5)

# Each chip has to have a definition so the script knows how to find it

ATtiny441 = {
            "name": "ATtiny441",
            "sig": [0x1E, 0x92, 0x15],
            "flash_size": 4*1024,
            "page_size": 16,
            "fuse_mask": (0xAE, 0xDF, 0xF5, 0x03),
            "clock_speed": 100000,
        }
attiny441 = ATtiny441

def error(err):
    """Helper to print out errors for us and then halt"""
    print("ERROR: " + err)
    avrprog.end()
    while True:
        pass

while input("Ready to GO, type 'G' here to start> ") != "G":
    pass

if not avrprog.verify_sig(attiny441, verbose=True):
    error("Signature read failure")
print("Found", attiny441["name"])

avrprog.write_fuses(attiny441, low=0x7A, high=0xFF)
if not avrprog.verify_fuses(attiny441, low=0x7A, high=0xFF):
    error("Failure verifying fuses!")

print("Programming flash from file")
avrprog.program_file(attiny441, "main.hex", verbose=True, verify=True)

print("Done!")
```

sebastianneubauer, 2023-02-27:
Wohoo, this is just awesome! I found a Atmel AVRIPS MKII, no clue why I have such a thing...but it actually worked and I uploaded your main.hex file to the ATTiny (after figuring out, that I need to power the attiny externally): `avrdude -c avrispmkII -P usb -p t441 -v -U flash:w:main.hex`
...and the sensor is up and running again!! Just it seems, the light measurement always shows 65535, is it the right firmware for the 2.7.1 version? Thank you so much!