Miceuz/i2c-moisture-sensor #42: Reading capacitance fails on esp32 after sleeping

ethanhjennings, 2023-06-11:
Hi, loving the sensor and this library but I can't get seem to get the `sleep()` method to work on an esp32. After sleeping all subsequent returned values are 65535 and I get an i2c error: `requestFrom(): i2cRead returned Error 263` until I do a hard reset by cutting power to the sensor.

Example code with issue:

```cpp
Wire.begin(I2C_SDA_PIN, I2C_SCL_PIN);
Wire.setClock(50000);

I2CSoilMoistureSensor moistureSensor;
moistureSensor.begin(true);

while (true) {
    auto moisture = moistureSensor.getCapacitance();
    Serial.println(moisture);
    
    moistureSensor.sleep();
    delay(1000);
}
```

Output:
```
258
[  4099][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
65535
[  8122][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
65535
[ 12145][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
65535
[ 16167][E][Wire.cpp:513] requestFrom(): i2cRead returned Error 263
65535
```

When I remove the `moistureSensor.sleep()` line it works fine though.

Miceuz, 2023-06-11:
Hi, thanks for the report,

What version of ESP32 are you using? This might be related to https://github.com/espressif/esp-idf/issues/4173 - 13ms max clock stretching limit on older ESP32 chip versions.

ethanhjennings, 2023-06-12:
Hi, I think this is indeed the problem because I'm using a regular ESP32 (not ESP32-S2 or S3). I'm going to try working around this with bit banging or a software i2c library (maybe then switching to using hardware i2c once it's awake).

closed 2023-06-12 by ethanhjennings